RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #1: Jetson Nano

wgthompson, 2020-07-09:
Hello,
Thank you for this driver we have all been so anxiously waiting for.

Is this compatible with the jetson nano, and if yes, would implementation be any different?
Thanks,
Will

rrcarlosrodriguez, 2020-07-09:
Hi @wgthompson , current patch works for Xavier NX. For Jetson Nano the driver should be compatible but the DTB might need some changes, but we haven't tested yet. We'll do next week and will release a new patch for Nano if needed. I will let you know once it is tested and available

wgthompson, 2020-07-09:
Cool, Thanks !

rrcarlosrodriguez, 2020-07-27:
Hi @wgthompson , we just updated the repository including Jetson Nano support. Please check and let us know if if works for you

closed 2020-08-06 by DavidSoto