RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #2: Arducam IMX477 camera

razvanphp, 2020-07-16:
Thank you very much for open-sourcing this, very much appreciated!

We have bough some of [those](https://www.uctronics.com/arducam-high-quality-camera-for-jetson-nano-12-3mp-1-2-3-inch-imx477-hq-camera-module-with-6mm-cs-mount-lens.html) cameras to test, but the jetvariety board/driver they provide does not support gstreamer and/or built-in ISP as discussed here: https://github.com/ArduCAM/MIPI_Camera/issues/44#issuecomment-658848960.

Would be very nice to adapt this camera for this driver, maybe they are not so different, since it has all 4 lanes connect and it can do very high fps:
Full resolution@60fps(normal), 4K2K@60fps, 1080p@240fps

We can sponsor the camera hardware and help with jetson testing, just let me know.

rrcarlosrodriguez, 2020-07-16:
Hi @razvanphp , we would be happy to help you supporting those cameras on Jetson ! Please send an email to support@ridgerun.com and we will process your inquiry and discuss next steps.

razvanphp, 2020-07-17:
We did send you an email days back, we tried to call multiple times... nobody answers.

It seems arducam already adapted and launched this driver for their camera, but on the jetson nano platform. Only thing left is allowing more FPS, since that camera has all 4 lanes connected, can we just adjust the code where max fps is defined or more stuff needs to be changed?

rrcarlosrodriguez, 2020-07-17:
@razvanphp , sounds weird, we didn't received a email from you, to which email did you send your request? And what days? So we can take a look to our inbox

We have received several emails from people asking for support on this topic and we have tried to do our best supporting everyone.

We don't have the arducam and we don't know the hardware, but by adjusting the DTB to support more FPS is not the only thing to do, more changes need to be applied to the driver and sensor configuration if you want to extend the support to 4 lanes and more FPS

razvanphp, 2020-07-20:
@rrcarlosrodriguez 

Sent: Monday, July 13, 2020 4:46 PM
To: 'support@ridgerun.com' <support@ridgerun.com>
Subject: Digital image stabilization

Can we talk in a conference or on the phone?

But we are also interested in adjusting the driver for higher FPS, we can send you the arducam or you can buy it yourself, where is the RD team located?

rrcarlosrodriguez, 2020-07-20:
@razvanphp got it, I found it, sorry for the late reply. Let's discuss details over email!

closed 2020-07-24 by rrcarlosrodriguez