RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #3: JetPack 4.4 – L4T R32.4.3 is now available

razvanphp, 2020-07-16:
Should the patch work with 4.3 revision too (released 07.07.2020), or only with JetPack 4.4 DP – L4T R32.4.2 (Developer Preview)? Maybe this should be mentioned in the wiki too. Thank you!

PS: Nvidia has the most confusing versioning ever seen.

rrcarlosrodriguez, 2020-07-17:
Since this first version of the driver works for NX only, it would work on JP 4.4 only. We will work in following weeks to add the required changes to support Nano, and it could work on JP 4.3.

razvanphp, 2020-07-18:
Jetson Nano also works on JetPack 4.4 now, 4.3 is the revision, not JetPack version, see [here](https://developer.nvidia.com/embedded/linux-tegra):

```
NVIDIA L4T 32.4.3 supports Jetson AGX Xavier series, Jetson Xavier NX, Jetson TX2 series, Jetson TX1, and Jetson Nano. It is included as part of JetPack 4.4 Production Release.
```

What I meant is that there are 2 variants of JP 4.4 now, DP (Developer Preview) and just 4.4, which is the Production Release, see the screenshot attached.
[image]

DavidSoto, 2020-07-20:
Hi, thanks for the feedback! we added a section for that. 
https://developer.ridgerun.com/wiki/index.php?title=Raspberry_Pi_HQ_camera_IMX477_Linux_driver_for_Jetson#Download_the_JetPack_4.4

We look forward to adding support for other platforms. We will keep you posted with the progress.

closed 2020-07-20 by DavidSoto