RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #4: Driver loaded, but cameras not detected

jchillerup, 2020-07-24:
I've followed the [guide on the wiki](https://developer.ridgerun.com/wiki/index.php?title=Raspberry_Pi_HQ_camera_IMX477_Linux_driver_for_Jetson) to apply the kernel patches and build the kernel for the Raspberry Pi IMX477 module. In the flashing step the authors have used the `-r` option to `flash.sh` to reuse a previously built `system.img` which I didn't have. So I built one using this command:

```
sudo ./flash.sh --no-flash -d kernel/dtb/tegra194-p3668-all-p3509-0000.dtb jetson-xavier-nx-devkit mmcblk0p1
```

That creates the `system.img` as I needed, and I verified that I was able to flash it to my NX devevelopment board. I could then proceed to flashing the DTB, and it booted. `dmesg` mentions imx477 so I guess I'm in the clear. (Intermission: then I `apt upgrade`d, overwrote the kernel and had a bitch of a fight to get it booting the proper kernel and proper DTB again, but now I'm back.)

I have the problem that the Argus camera driver does not detect the camera. This is the output from gstreamer...

```root@jc-nx:~# gst-launch-1.0 -e nvarguscamerasrc sensor-id=$SENSOR_ID ! "video/x-raw(memory:NVMM),width=1920,height=1080,framerate=$FRAMERATE/1" ! nvv4l2h264enc ! h264parse ! mp4mux ! filesink location=rpi_v3_imx477_cam$SENSOR_ID.mp4
nvbuf_utils: Could not get EGL display connection
nvbuf_utils: ERROR getting proc addr of eglCreateImageKHR
nvbuf_utils: ERROR getting proc addr of eglDestroyImageKHR
Setting pipeline to PAUSED ...
Opening in BLOCKING MODE 
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Redistribute latency...
NvMMLiteOpen : Block : BlockType = 4 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4 
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:557 No cameras available
Got EOS from element "pipeline0".
Execution ended after 0:00:00.041150301
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
(Argus) Error EndOfFile: Unexpected error in reading socket (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 266)
(Argus) Error EndOfFile: Receive worker failure, notifying 1 waiting threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 340)
(Argus) Error InvalidState: Argus client is exiting with 1 outstanding client threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 357)
(Argus) Error EndOfFile: Receiving thread terminated with error (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadWrapper(), line 368)
(Argus) Error EndOfFile: Client thread received an error from socket (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 145)
(Argus) Error EndOfFile:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 87)```

How would I proceed to debug this? I have removed the R8 as directed by the guide, so this should not be a power problem.

Thanks!

jchillerup, 2020-07-24:
[Here's the dmesg](https://termbin.com/oj7g)

jchillerup, 2020-07-24:
I installed the ISP settings, which didn't help either :(

jchillerup, 2020-07-24:
Following [this link](https://forums.developer.nvidia.com/t/gstreamer-cannot-capture-from-bayer-sensor-imx415-in-jetson-nano/139674/4) it seems something *is* coming out of the camera after all...

```
root@jc-nx:~# v4l2-ctl --stream-mmap --stream-count=100 -d /dev/video0
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.09 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.04 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.03 fps
```

razvanphp, 2020-07-24:
`export SENSOR_ID=0` ?

rrcarlosrodriguez, 2020-07-24:
Do you have the kernel modules properly loaded? What is the output of `lsmod` ?

jchillerup, 2020-07-24:
Well, no. In fact there were no modules loaded. It seems the kernel was looking for modules in `/lib/modules/4.9.140` but there was no such directory, rather `/lib/modules/4.9.140-tegra`. I made a symlink, and now it works! Thanks @rrcarlosrodriguez

closed 2020-07-24 by jchillerup