RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #5: Jetson Nano I2C Failure registering /cam

AxelZschutschke, 2020-07-30:
# driver issue with Jetson Nano

## approach
Test kernel patch with 2x IMX477 Raspberry HQ cams on Jetson Nano and Jetson Xavier
-> Xavier works fine
-> Nano does not detect cams

## issue observed
While working fine with Jetson Xavier, when switching to Nano, the same hardware-arrangement except for the Jetson module, gives me error message:
~~~
gst-launch-1.0 -e nvarguscamerasrc num-buffers=$NUMBER_OF_SNAPSHOTS sensor-id=$SENSOR_ID ! "video/x-raw(memory:NVMM),width=1920,height=1080,framerate=$FRAMERATE/1" ! nvjpegenc ! multifilesink location=%03d_rpi_v3_imx477_cam$SENSOR_ID.jpeg
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:557 No cameras available
Got EOS from element "pipeline0".
Execution ended after 0:00:00.214653860
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
~~~

## environment

Jetson Nano B01 on dev-kit board B01 with JetPack 4.4

kernel has been custom build following procedure description [ridgerun](https://developer.ridgerun.com/wiki/index.php?title=Raspberry_Pi_HQ_camera_IMX477_Linux_driver_for_Jetson#Copy_the_kernel_modules)
~~~
uname -a
Linux jetson-nano-b01 4.9.140 #1 SMP PREEMPT Thu Jul 30 11:19:59 CEST 2020 aarch64 aarch64 aarch64 GNU/Linux
~~~

~~~
lsmod
Module                  Size  Used by
bnep                   19334  2
fuse                  120331  3
zram                   30209  4
overlay                54990  0
spidev                 14763  0
nvgpu                1734448  18
imx477                 19780  0
bluedroid_pm           16187  0
ip_tables              21475  0
x_tables               38144  1 ip_tables
~~~

~~~ 
dmesg | grep "imx\|ic2\|video"
[    0.000000] Kernel command line: tegraid=21.1.2.0.0 ddr_die=4096M@2048M section=512M memtype=0 vpr_resize usb_port_owner_info=0 lane_owner_info=0 emc_max_dvfs=0 touch_id=0@63 video=tegrafb no_console_suspend=1 console=ttyS0,115200n8 debug_uartport=lsport,2 earlyprintk=uart8250-32bit,0x70006000 maxcpus=4 usbcore.old_scheme_first=1 lp0_vec=0x1000@0xff780000 core_edp_mv=1075 core_edp_ma=4000  earlycon=uart8250,mmio32,0x70006000  root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyS0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0 quiet
[    0.437496] iommu: Adding device 546c0000.i2c to group 22
[    0.438874] tegra-pmc 7000e400.pmc: i2c-thermtrip node not found, emergency thermal reset disabled.
[    0.490255] iommu: Adding device 7000c000.i2c to group 25
[    0.490494] iommu: Adding device 7000c400.i2c to group 26
[    0.490740] iommu: Adding device 7000c500.i2c to group 27
[    0.490981] iommu: Adding device 7000c700.i2c to group 28
[    0.491247] iommu: Adding device 7000d000.i2c to group 29
[    0.491487] iommu: Adding device 7000d100.i2c to group 30
[    0.594897] Linux video capture interface: v2.00
[    1.710755] i2c /dev entries driver
[    1.712835] i2c i2c-7: Failed to register i2c client imx477 at 0x1a (-16)
[    1.719677] i2c i2c-7: of_i2c: Failure registering /cam_i2cmux/i2c@0/rbpcv3_imx477_a@1a
[    1.727697] i2c i2c-7: Failed to create I2C device for /cam_i2cmux/i2c@0/rbpcv3_imx477_a@1a
[    1.727723] i2c i2c-6: Added multiplexed i2c bus 7
[    1.728072] i2c i2c-8: Failed to register i2c client imx477 at 0x1a (-16)
[    1.734880] i2c i2c-8: of_i2c: Failure registering /cam_i2cmux/i2c@1/rbpcv3_imx477_e@1a
[    1.742893] i2c i2c-8: Failed to create I2C device for /cam_i2cmux/i2c@1/rbpcv3_imx477_e@1a
[    1.742912] i2c i2c-6: Added multiplexed i2c bus 8
[    1.742920] i2c-mux-gpio cam_i2cmux: 2 port mux on Tegra I2C adapter adapter
[    3.999112] imx477 6-001a: tegracam sensor driver:imx477_v2.0.6
~~~

~~~
cat /boot/extlinux/extlinux.conf 
TIMEOUT 30
DEFAULT primary

MENU TITLE L4T boot options

LABEL primary
      MENU LABEL primary kernel
      LINUX /boot/Image
      INITRD /boot/initrd
      FDT /boot/tegra210-p3448-0000-p3449-0000-b00.dtb
      APPEND ${cbootargs} quiet
~~~

device tree file seems active, /proc/device-tree shows imx477 nodes (all 3 of them in host1x and cami2cmux)
~~~
cat  /proc/device-tree/cam_i2cmux/i2c@*/rbpcv3_imx477_*/devnode
video0video1
~~~

AxelZschutschke, 2020-07-30:
Any help highly appreciated. I am happy to test and/or supply additional info!

AxelZschutschke, 2020-07-30:
Side-note 1: /dev does not have any "video" listings
Side-note 2: using your kernel driver but the device-tree file  tegra210-p3448-0000-p3449-0000-b00.dtb from the [arducam-guys](https://github.com/ArduCAM/MIPI_Camera/tree/master/Jetson/IMX477/driver/Nano) seemingly works... unfortunately I did not find any sources from them

AxelZschutschke, 2020-07-30:
Checked out dts diffs after de-compiling: only found a couple of renames and the following diffs:
you use **status : "okay"** -> setting "disabled" seems to fix the issue (or shadows) it - but I am able to capture images

zip contains the result of your patch and a "fixed" version, which does the trick for me
[decompiled_dts.tar.gz](https://github.com/RidgeRun/NVIDIA-Jetson-IMX477-RPIV3/files/5001659/decompiled_dts.tar.gz)

AxelZschutschke, 2020-07-31:
Did some more tests with the workaround image today, cams are working but driver is refusing cam id #1. Seems it does not allow for "compatible":"nvidia,imx477" -> I changed this to "riderun/imx477" as with the id #0 cam and now it works...

ClonedPuppy, 2020-08-01:
Same issue here. i2c fails to register. 
```
peter@ariel:~$ dmesg | grep "imx\|ic2\|video"
[    0.000000] Kernel command line: tegraid=21.1.2.0.0 ddr_die=4096M@2048M section=512M memtype=0 vpr_resize usb_port_owner_info=0 lane_owner_info=0 emc_max_dvfs=0 touch_id=0@63 video=tegrafb no_console_suspend=1 console=ttyS0,115200n8 debug_uartport=lsport,2 earlyprintk=uart8250-32bit,0x70006000 maxcpus=4 usbcore.old_scheme_first=1 lp0_vec=0x1000@0xff780000 core_edp_mv=1075 core_edp_ma=4000 tegra_fbmem=0x800000@0x92cb0000 is_hdmi_initialised=1  root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyS0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0    root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyS0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0 quiet
[    0.619555] Linux video capture interface: v2.00
[    1.210729] i2c i2c-7: Failed to register i2c client imx477 at 0x1a (-16)
[    1.218354] i2c i2c-7: of_i2c: Failure registering /cam_i2cmux/i2c@0/rbpcv3_imx477_a@1a
[    1.226402] i2c i2c-7: Failed to create I2C device for /cam_i2cmux/i2c@0/rbpcv3_imx477_a@1a
[    1.226678] i2c i2c-8: Failed to register i2c client imx477 at 0x1a (-16)
[    1.233502] i2c i2c-8: of_i2c: Failure registering /cam_i2cmux/i2c@1/rbpcv3_imx477_e@1a
[    1.233506] i2c i2c-8: Failed to create I2C device for /cam_i2cmux/i2c@1/rbpcv3_imx477_e@1a
[    1.233926] imx477 6-001a: tegracam sensor driver:imx477_v2.0.6
[    1.257287] imx477 6-001a: imx477_board_setup: error during i2c read probe (-121)
[    1.257313] imx477 6-001a: board setup failed
[    1.257374] imx477: probe of 6-001a failed with error -121
```
Cheers,

Peter.

rrcarlosrodriguez, 2020-08-04:
Hi @AxelZschutschke, we just pushed some changes to the repo. Give it a try and let me know if it fixes your issue. In Jetson Nano only 1 camera should load. You can see /dev/video0 device after loading the kernel and DTB

AxelZschutschke, 2020-08-04:
Hi @rrcarlosrodriguez, thank you for the support and the good work alltogether! I will try the version during the day. But one question straight away: why only one camera?

rrcarlosrodriguez, 2020-08-04:
Hi @AxelZschutschke , interesting, after reading your question I was wondering why people is talking about 2 cameras support in the Jetson Nano. And then I realized we are working with the very first model, A02, which has 1 camera port only. There is a new EV Kit model, B01 which has 2 camera ports, which is probably the one you have. You can see the differences [here](https://www.seeedstudio.com/blog/2020/01/16/new-revision-of-jetson-nano-dev-kit-now-supports-new-jetson-nano-module/) 

Please test the new version of the patches and let me know if it works with the 2 cameras in the Jetson Nano.

ClonedPuppy, 2020-08-05:
Just tried the latest changes and it still fails to register the i2c client for me.. It's basically the same error msg as before. I'm also running the B01 board with two camera outputs, my camera is in the cam0 port.

zinuok, 2020-08-05:
Even if I'm using Xavier NX (not Nano), it still fails just like @ClonedPuppy ..
How did you make that in NX, @AxelZschutschke  ?

juankruz, 2020-08-05:
Hi @zinuok. Did you perform any hardware change? 
We didn't see problems with i2c buses and your problem happens like if our hardware was different.

-JC

DavidSoto, 2020-08-05:
Just to be clear, to get the camera in the i2c bus you need to make this change:

https://developer.ridgerun.com/wiki/index.php?title=Raspberry_Pi_HQ_camera_IMX477_Linux_driver_for_Jetson#Compatibility_with_NVIDIA.C2.AEJetson.E2.84.A2_Platforms

Did you do the change already?

ClonedPuppy, 2020-08-06:
@DavidSoto , not sure if you're asking me or @zinuok, but if me, then yep, mod has been done. I''m on a Nano however, B01 board (two cam ports).

zinuok, 2020-08-06:
To @juankruz and @DavidSoto :
my hardware is
- Xavier NX, Jetpack 4.4
- IMX477 (HQ camera V1.0 2018)
I didn't remove the resistor, since following link said we don't have to remove it.
https://www.youtube.com/watch?v=uIycS8989fg, which is based on the Ridgerun code.
So I followed the link DavidSoto said with the resistor.

should we remove the resistor in Xavier NX?

zinuok, 2020-08-06:
After removing the R8 resistor from IMX477, it works very well with Xavier NX.
Thank you for @juankruz and @DavidSoto , also @ClonedPuppy for inviting me to this issue
Thanks :)

DavidSoto, 2020-08-06:
Glad to hear that it worked!

rrcarlosrodriguez, 2020-08-14:
Hi,

We just pushed a new change in the DTB, to enable 2 cameras on the Jetson Nano revision B01.This new version of the patches fixes this issue. Closing it now. Feel free to let us know how it works for you

closed 2020-08-14 by rrcarlosrodriguez