RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #6: Patch fails

ClonedPuppy, 2020-07-31:
Hello, not being very successful in applying the patch. Tried a few times on a vanilla install of 4.4 DP. Here is my log output when the patch fails:

`peter@UbuntuExternal:~/nvidia/nvidia_sdk/JetPack_4.4_DP_Linux_DP_JETSON_NANO_DEVKIT/Linux_for_Tegra/source$ quilt push
Applying patch patches/4.4.3_l4t32.4.3_rbpv3_imx477_support.patch
patching file hardware/nvidia/platform/t19x/jakku/kernel-dts/common/tegra194-camera-jakku-rbpcv3-imx477.dtsi
patching file hardware/nvidia/platform/t19x/jakku/kernel-dts/common/tegra194-camera-rbpcv3-imx477.dtsi
patching file hardware/nvidia/platform/t19x/jakku/kernel-dts/common/tegra194-p3509-0000-a00.dtsi
patching file kernel/nvidia/drivers/media/i2c/Kconfig
patching file kernel/nvidia/drivers/media/i2c/Makefile
patching file kernel/nvidia/drivers/media/i2c/imx477.c
patching file kernel/nvidia/drivers/media/i2c/imx477_mode_tbls.h
patching file kernel/nvidia/include/media/imx477.h
patching file hardware/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-camera-rbpcv3-dual-imx477.dtsi
patching file hardware/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-camera-rbpcv3-imx477.dtsi
patching file hardware/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-porg-camera-rbpcv3-dual-imx477.dtsi
patching file hardware/nvidia/platform/t210/porg/kernel-dts/porg-platforms/tegra210-porg-camera-rbpcv3-imx477.dtsi
can't find file to patch at input line 3077
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|Index: sources/hardware/nvidia/platform/t210/porg/kernel-dts/tegra210-porg-p3448-common.dtsi
|===================================================================
|--- sources.orig/hardware/nvidia/platform/t210/porg/kernel-dts/tegra210-porg-p3448-common.dtsi
|+++ sources/hardware/nvidia/platform/t210/porg/kernel-dts/tegra210-porg-p3448-common.dtsi
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
can't find file to patch at input line 3092
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|Index: sources/hardware/nvidia/platform/t210/porg/kernel-dts/porg-plugin-manager/tegra210-porg-plugin-manager.dtsi
|===================================================================
|--- sources.orig/hardware/nvidia/platform/t210/porg/kernel-dts/porg-plugin-manager/tegra210-porg-plugin-manager.dtsi
|+++ sources/hardware/nvidia/platform/t210/porg/kernel-dts/porg-plugin-manager/tegra210-porg-plugin-manager.dtsi
--------------------------
No file to patch.  Skipping patch.
16 out of 16 hunks ignored
Patch patches/4.4.3_l4t32.4.3_rbpv3_imx477_support.patch does not apply (enforce with -f)
`
Any ideas?

Thanks.

Peter.

juankruz, 2020-07-31:
Hi @ClonedPuppy.

We are currently performing some changes on this repository structure by placing the sources of Xavier NX and Nano in different folders. We will let you know once the modifications are done on today afternoon.

Thank you,

JC

AxelZschutschke, 2020-07-31:
But it works if you just force the patch (quilt push -f) as the nano and the xavier use different paths for the device-tree sources...

DavidSoto, 2020-07-31:
Right, this is why we will create two directories, one per platform. In order to avoid these problems. We will have it ready by Monday.

ClonedPuppy, 2020-08-01:
> Hi @ClonedPuppy.
> 
> We are currently performing some changes on this repository structure by placing the sources of Xavier NX and Nano in different folders. We will let you know once the modifications are done on today afternoon.
> 
> Thank you,
> 
> JC

Thank you JC, looking forward to the new structure.

ClonedPuppy, 2020-08-01:
> But it works if you just force the patch (quilt push -f) as the nano and the xavier use different paths for the device-tree sources...

I just went ahead and did a -f as well, but once i get to the step where i need to copy out the kernel 
`cp build/arch/arm64/boot/dts/tegra194-p3668-all-p3509-0000.dtb ./kernel/dtb`
It fails, as I don't have that **tegra194-p3668-all-p3509-0000.dtb** built in my dts directory. Mine are all tegra210*, is that another Nano particularity as well?

ClonedPuppy, 2020-08-01:
Sorry, figured out it had to be the **tegra210-p3448-0000-p3449-0000-b00.dtb** file by reading the below instructions a bit more closely. Successfully rebooted, will try removing the resistor and hooking up the camera now.

juankruz, 2020-08-03:
Hi @ClonedPuppy and @AxelZschutschke.

We have modified the repository structure and also updated the driver installation instructions.
Please, give it a try and let us know how your testing goes.

JC

ClonedPuppy, 2020-08-05:
Patch works perfect on the Nano now JC, thanks a lot.

closed 2020-08-05 by ClonedPuppy