RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #7: Flahs failed: Invalid target board - jetson-xavier-nx-devkit

zinuok, 2020-08-03:
I'm trying to flash my NX board, which is the last step for using IMX477.
If I write
$ sudo ./flash.sh -r -k kernel-dtb -d source/kernel/dtb/tegra194-p3668-all-p3509-0000.dtb jetson-xavier-nx-devkit mmcblk0p1

this error occurs:
Error: Invalid target board - jetson-xavier-nx-devkit.

I did googling and found that target board name is correct..
Is there any solution?

( following is my tree sturcture: ~/nvidia/nvidia_sdk/JetPack_4.4_DP_Linux_DP_JETSON_XAVIER_NX_DEVKIT/Linux_for_Tegra)

.
├── build
├── dtb
├── flash.sh
├── modules
├── packages
└── source

zinuok, 2020-08-03:
According to this link: https://forums.developer.nvidia.com/t/cant-flash-tx2-via-jetpack-invalid-target-board/50034
the error was due to a missing symlink that jetpack somehow didn’t create.

So I think this is a just dependency problem, but I can't understand
where did you guys get the 'flash.sh' file?
Or could you upload the flashed image file?

ClonedPuppy, 2020-08-03:
It should be inside your host machine's nvidia sdk directory, in your case since you are using the xavier, I think it should be JetPack_4.4_Linux_JETSON_XAVIER_NX_DEVKIT or JetPack_4.4_DP_Linux_DP_JETSON_XAVIER_NX_DEVKIT (if using the developer preview). The flash.sh script is inside the Linux_for_Tegra folder in the above diretory.

ClonedPuppy, 2020-08-03:
Reading your first post more closely, are you sure you are not mixing the two? the non DP version and the "regular" jetpack version? I'm not sure if that would work.

zinuok, 2020-08-03:
Thank you for your kindness.

* As like your comment, I took the 'flash.sh' file from JetPack_4.4_Linux_JETSON_XAVIER_NX_DEVKIT directory, since the etPack_4.4_DP_Linux_DP_JETSON_XAVIER_NX_DEVKIT directory(used in IMX477 patch process) didn't include it.

* What is the difference between DP version and "regular" jetpack version ?
(I guess they are not mixed, since I followed the ref. link step by step carefully..)

zinuok, 2020-08-03:
The weird thing is
In the last step of **"Build kernel"**, the insturction tries to access "kernel" directory in the 'nvidia_sdk/JetPack_4.4_DP_Linux_DP_JETSON_XAVIER_NX_DEVKIT/Linux_for_Tegra', which doesn't exist as the tree structure in my first post 
____
Copy the generated kernel and dtb:
cd $JETSON_KERNEL_SOURCE/../
cp build/arch/arm64/boot/dts/tegra194-p3668-all-p3509-0000.dtb ./kernel/dtb
cp build/arch/arm64/boot/Image ./kernel
____

I suspect the DP version didn't installed properly

ClonedPuppy, 2020-08-03:
The DP version stands for a Developer Preview, sort of a beta version before official release. But as far as I understand, according to Ridgerun, the driver will work with either, as the kernel source tree should be the same. I was successful patching it with both, but that was on a Nano device.  

If I were you, I would open up the sdkmanager, and then uninstall everything (go the the small repair/uninstall link on the app in the left corner). After that, close it, and delete any nvidia folder you might have in ~/Downloads and ~/. Then do a complete retry of it all. Finger crossed and it will work ;)

zinuok, 2020-08-03:
Thank you for your advice.
Until now I have installed the original Jetpack version, not 'DP' version you mentioned by sdk manager
So I'm gonna try to the instructions you said with the 'DP' version

Thank you very much
sincerely :)

rrcarlosrodriguez, 2020-08-04:
Thanks for the help @ClonedPuppy . We have confirmed the patch works for both, JP 4.4 versions. Closing this issue

closed 2020-08-04 by rrcarlosrodriguez