RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #8: Flashing issue: Xavier NX

zinuok, 2020-08-04:
If we do flash the xavier NX (copy Image into NX /boot/ directory), 
we don't have to do 'Copy the kernel modules' right?
I tried to copy the kernel modules following the instruction here. However I failed that since NX is already out of memory after initial setup.

After copying the Image file, I did reboot my NX but it falls in bootloop infinitely..
anyone succeed flahsing after this repository being updated yesterday?

AxelZschutschke, 2020-08-04:
Didn't try it yesterday, but other Linux Systems / Kernel images are very sensitive regarding the module build version / compiler / environment used for kernel build. So a custom build kernel image might as well refusing the stock kernel modules. For storage reasons, you can dump the original /lib/modules/4.... folder and copy the new custom build modules there (without rebooting inbetween) Make sure the new folder is named exactly as in your build tree (dont add anything to the module folder name while copying)
e.g. original was named 4.9.140-tegra in my case, I moved it away and placed 4.9.140 (without the tegra) as my custom build used that path.

zinuok, 2020-08-05:
thank you for reply. I'll try that

I have a question:
The last step of this repository is "Flashing", but the ridgerun site also includes "copy the kernel module". Should we do that?

AxelZschutschke, 2020-08-05:
Again, did not try without but I would assume yes - you need to do that. Also the custom build "Image" (Kernel) probably searches /lib/modules/4.9.140 (or whatever version you are using), which probably is not present in your case without copying the modules there? Please make sure the /lib/module/XXX path (the XXX part) is exactly the same as in your build folder...

zinuok, 2020-08-05:
Thank you!

Sorry to bother you, but I can't find any instruction about "removing symlink".

Does
'make sure to remove the build and source symlinks before copying the modules (as written in the ridgerun instructions!'
means that we don't need to copy ../4.9.140/build and source directory to Xavier NX?

the symlink build points to '../Linux_for_Tegra/build', which is 7.1GB! ( I can't copy this even if I dump original one in Xavier NX as you said)

I'm confused cause the built files are need to boot up the Xavier NX

Thank you

ClonedPuppy, 2020-08-05:
Hey @zinuok , I think we are in similar timezones so I'll jump in and answer for @AxelZschutschke :)
You can just disregard the build and source directories, you don't need those. Just the modules, it quite large however, close to 1gb. If you used the SDKmanager style of flashing the Nano then your storage is max 16Gb, which is a real pain in the #$%!. What you can do if you need more space is to run a script which will fix the partitions for you, in case you use a larger SD card. Let me know if you need to do that.

So regarding the modules, yes it's a good idea to copy them over. Most likely they will probably be identical to the ones that were there already, but just to be safe. The only difference that we know of, is that the stock kernel that is installed by nvidia, looks for the kernels in a 4.9.140-tegra directory, while the one we built looks for it in the 4.9.140 directory (no -tegra). If you feel adventurous you could create a symlink from your stock 4.9.140-tegra to a 4.9.140 and do a reboot. I guess theoretically it should work, since we didn't really add or remove any of the modules in menuconfig... but again, no guarantees.

zinuok, 2020-08-05:
Thank you for your help to @ClonedPuppy  and @AxelZschutschke 
the cause of my problem was I copied as '4.9.140-tegra', not '4.9.140'

Now I can reboot successfully my NX board, but still my board can't recognize IMX477 with this message: "Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:557 No camera available"

Did you guys have this kinds of problem?
I'm trying to work out that problem.. but it's not easy

ClonedPuppy, 2020-08-05:
Check this [issue](https://github.com/RidgeRun/NVIDIA-Jetson-IMX477-RPIV3/issues/5), if same, then welcome to the club :D

zinuok, 2020-08-06:
> If we do flash the xavier NX copying Image into NX /boot/ directory,
> we don't have to do 'Copy the kernel modules' right?
> I tried to copy the kernel modules following the instructions here. However I failed that since NX is already out of memory after initial setup.
> 
> After copying the Image file, I did reboot my NX but it falls in bootloop infinitely..
> anyone succeed flahsing after this repository being updated yesterday?

dd

zinuok, 2020-08-06:
close this issue, since I solved it by removing R8 resistor. Refer this: #5

closed 2020-08-06 by zinuok