RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #10: Jetson Nano Error at Compile kernel, device tree and modules:

nr-patel, 2020-08-11:
I am getting following error after running:
make -C kernel/kernel-4.9/ ARCH=arm64 O=$KERNEL_OUT CROSS_COMPILE=${CROSS_COMPILE} -j4 Image
Error:arch/arm64/Makefile:49: LSE atomics not supported by binutils

rrcarlosrodriguez, 2020-08-12:
Hi @nr-patel , just a quick question, are you running the compilation procedure in your host machine or inside the Jetson Nano?

nr-patel, 2020-08-12:
Hello @rrcarlosrodriguez ,

I have already setup my Jetson Nano. I am just following steps from option B to add driver patch. When will be option A steps will be available for Jetson Nano.

Thanks,
Nikhil

rrcarlosrodriguez, 2020-08-12:
Compilation procedure should be done inside you host PC, not inside the Jetson Nano.

Option A will be available next week. We already received the Jetson Nano B01 version and will be working on it this week

closed 2020-09-17 by rrcarlosrodriguez