RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #9: Updated ISP file gives worse quality

fhmattsson, 2020-08-11:
The updated ISP file gives worse quality  the image

- White spot flickering pixels
- Image quality is reduced, big difference if you change the De-noising level

rrcarlosrodriguez, 2020-08-12:
Hi @fhmattsson ,

Thanks for your feedback, we'll check with Leopard Imaging about this feedback and get back to you once we got an answer.

juankruz, 2020-08-13:
Hi @fhmattsson.

We have not seen the undesired behavior you mentioned.
Would you be able to share some captures or short videos showing us that effects?

Please, also attach the .isp file and the lens model you are using.

Thank you!

JC

fhmattsson, 2020-11-28:
Hi, 
Sorry for a long delay.

Here's two files New & Old, with a cropped region where the difference in the ISP's are obvious

I'm using a 12MP 5mm lens

/Fredrik

[ISP Control.zip](https://github.com/RidgeRun/NVIDIA-Jetson-IMX477-RPIV3/files/5610230/ISP.Control.zip)

[image]
[image]
[image]
[image]