RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #12: how to change camera index?

AndreV84, 2020-08-21:


rrcarlosrodriguez, 2020-09-17:
Hi @AndreV84 ,

Could you please elaborate on the problem you are facing?

AndreV84, 2020-09-18:
Hi @ronnyjimenez 
I was trying to assist some folk using ardusimple HQ camera withh your drivers;
their concern was to make the browser [ mozilla] to use default camera . The idea was that it will be not the original camera, but an augmented stream on loopbackk like /dev/video2. HHowever, the firefox by default will always open /dev/video0;
thus the question emerged how to make the augmented loopback camera to be recognized as /dev/video0, and the original camera swap get to /dev/ video3  example

rrcarlosrodriguez, 2020-09-18:
@AndreV84 there is no straightforward to do. It is controlled by the internal VI drivers. For example, look at this diff:

`
diff --git a/kernel/nvidia/drivers/media/platform/tegra/camera/vi/channel.c b/kernel/nvidia/drivers/media/platform/tegra/camera/vi/channel.c
index 940b6bac3..5d15399a7 100644
--- a/kernel/nvidia/drivers/media/platform/tegra/camera/vi/channel.c
+++ b/kernel/nvidia/drivers/media/platform/tegra/camera/vi/channel.c
@@ -2255,9 +2255,6 @@ int tegra_channel_init(struct tegra_channel *chan)
        chan->video.fops = &tegra_channel_fops;
        chan->video.v4l2_dev = &vi->v4l2_dev;
        chan->video.queue = &chan->queue;
-       snprintf(chan->video.name, sizeof(chan->video.name), "%s-%s-%u",
-               dev_name(vi->dev), chan->pg_mode ? "tpg" : "output",
-               chan->pg_mode ? (chan->id - vi->num_channels) : chan->port[0]);
        chan->video.vfl_type = VFL_TYPE_GRABBER;
        chan->video.vfl_dir = VFL_DIR_RX;
        chan->video.release = video_device_release_empty;
diff --git a/kernel/nvidia/drivers/media/platform/tegra/camera/vi/graph.c b/kernel/nvidia/drivers/media/platform/tegra/camera/vi/graph.c
index 896c58103..e3b8b625d 100644
--- a/kernel/nvidia/drivers/media/platform/tegra/camera/vi/graph.c
+++ b/kernel/nvidia/drivers/media/platform/tegra/camera/vi/graph.c
@@ -455,6 +455,7 @@ int tegra_vi_get_port_info(struct tegra_channel *chan,
        struct device_node *ports;
        struct device_node *port;
        int value = 0xFFFF;
+       const char *temp_str;
        int ret = 0, i;
 
        ports = of_get_child_by_name(node, "ports");
@@ -488,6 +489,12 @@ int tegra_vi_get_port_info(struct tegra_channel *chan,
                                return -EINVAL;
                        }
 
+                       /* Get video device name */
+                       ret = of_property_read_string(ep, "devnode", &temp_str);
+                       if (ret < 0)
+                               dev_err(&chan->video.dev, "devnode error\n");
+                       strcpy (chan->video.name, temp_str);
+
                        /* Get CSI port */
                        ret = of_property_read_u32(ep, "port-index", &value);
                        if (ret < 0)
`

We are removing a line in the channel.c file, where the video device name is assigned, and then in the graph.c we are assigning the name based on the DTB specification, so after applying this change, you can add a `devnode = "video0"` to the DTB node, something like: 

`
port@0 {
                                        reg = <0>;
                                        test_vi_in0: endpoint {
+                                               devnode = "video1";
                                                port-index = <4>;
                                                bus-width = <4>;
`

Hope this helps, since this is not a IMX477 specific issue or feature I will close the issue. If you require further help you can get some professional support sending an inquiry to support@ridgerun.com

closed 2020-09-18 by rrcarlosrodriguez