RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #14: external sync signals support

ngeorgis, 2020-09-14:
thank you, great work! 

Any way to get the same support for sync as the pi guys have achieved in  

https://www.raspberrypi.org/forums/viewtopic.php?t=281913

`
Connect the XVS line on the HQ cameras together.
`
`On one Pi add "imx477_trigger_mode=1" to config.txt
`
`On any other Pis add "imx477_trigger_mode=2" to config.txt
`

rrcarlosrodriguez, 2020-09-15:
Hi @ngeorgis , this looks like a very customized change. User will need to apply it on their own for their use case. RidgeRun is not planning on adding that support

closed 2020-09-15 by rrcarlosrodriguez

DavidSoto, 2020-09-15:
Hi @ngeorgis, just to be clear, we have done similar modifications on drivers for other customers to put multiple cameras on sync on master/slave mode. However, this free version of the driver won't include it. If you are interested on hiring RidgeRun to help you adding this support please contact us support@ridgerun.com. You can read more about our services here:
https://developer.ridgerun.com/wiki/index.php?title=RidgeRun_Professional_Services

ngeorgis, 2020-09-15:
Thank you @rrcarlosrodriguez  and @DavidSoto !

ngeorgis, 2020-09-22:
> @ngeorgis, check out this RPi forum thread. https://www.raspberrypi.org/forums/viewtopic.php?t=281913

Thank you @fischer !  Yes, that was the original pi forum thread that inspired us to get same functionality for Jetson NX. Still need to read and understand a bit more.