RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #15: Error on Jetson Nano with Auvidea JN30B carrier board

mattcarp12, 2020-09-17:
I am successfully able to use the Arducam IMX477 with the Nvidia P449 carrier board, but receive an error when using the Auvidea JN30B carrier board, with the P3448-0002 production module.

After starting the sample gstreamer pipeline, I see the following logs in from `nvargus-daemon`:

```
-- Logs begin at Wed 2020-09-16 11:59:55 EDT, end at Wed 2020-09-16 12:18:58 EDT. --
Sep 16 11:59:59 monomer-auvidea-1 systemd[1]: Started Argus daemon.
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: === NVIDIA Libargus Camera Service (0.97.3)=== Listening for connections...=== gst-launch-1.0[7738]: Connection established (7F88C0F1D0)OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module0
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module1
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: NvPclHwGetModuleList: WARNING: Could not map module to ISP config string
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: NvPclHwGetModuleList: No module data found
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: NvPclHwGetModuleList: WARNING: Could not map module to ISP config string
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: NvPclHwGetModuleList: No module data found
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: OFParserGetVirtualDevice: NVIDIA Camera virtual enumerator not found in proc device-tree
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: ---- imager: Found override file [/var/nvidia/nvcam/settings/camera_overrides.isp]. ----
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: (NvCamV4l2) Error ModuleNotPresent: V4L2Device not available (in /dvs/git/dirty/git-master_linux/camera/utils/nvcamv4l2/v4l2_device.cpp, function findDevice(), line 256)
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: (NvCamV4l2) Error ModuleNotPresent:  (propagating from /dvs/git/dirty/git-master_linux/camera/utils/nvcamv4l2/v4l2_device.cpp, function initialize(), line 60)
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: (NvOdmDevice) Error ModuleNotPresent:  (propagating from dvs/git/dirty/git-master_linux/camera-partner/imager/src/devices/V4L2SensorViCsi.cpp, function initialize(), line 106)
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: NvPclDriverInitializeData: Unable to initialize driver v4l2_sensor
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: NvPclInitializeDrivers: error: Failed to init camera sub module v4l2_sensor
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: NvPclStartPlatformDrivers: Failed to start module drivers
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: NvPclStateControllerOpen: Failed ImagerGUID 1. (error 0xA000E)
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: NvPclOpen: PCL Open Failed. Error: 0xf
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: SCF: Error BadParameter: Sensor could not be opened. (in src/services/capture/CaptureServiceDeviceSensor.cpp, function getSourceFromGuid(), line 582)
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: SCF: Error BadParameter:  (propagating from src/services/capture/CaptureService.cpp, function addSourceByGuid(), line 437)
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: SCF: Error BadParameter:  (propagating from src/api/CameraDriver.cpp, function addSourceByIndex(), line 295)
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: SCF: Error BadParameter:  (propagating from src/api/CameraDriver.cpp, function getSource(), line 458)
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: Acquiring SCF Camera device source via index 0 has failed. ---- imager: Found override file [/var/nvidia/nvcam/settings/camera_overrides.isp]. ----
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: LSC: LSC surface is not based on full res!
Sep 16 12:04:09 monomer-auvidea-1 nvargus-daemon[5276]: === gst-launch-1.0[7738]: CameraProvider initialized (0x7f8491fda0)CAM: serial no file already exists, skips storing againLSC: LSC surface is not based on full res!
Sep 16 12:04:11 monomer-auvidea-1 nvargus-daemon[5276]: SCF: Error Timeout: ISP Stats timed out! (in src/services/capture/NvIspHw.cpp, function waitIspStatsFinished(), line 561)
Sep 16 12:04:11 monomer-auvidea-1 nvargus-daemon[5276]: SCF: Error Timeout: ISP port 0 timed out! (in src/services/capture/NvIspHw.cpp, function waitIspFrameEnd(), line 478)
Sep 16 12:04:11 monomer-auvidea-1 nvargus-daemon[5276]: SCF: Error Timeout:  (propagating from src/services/capture/NvIspHw.cpp, function waitIspFrameEnd(), line 519)
Sep 16 12:04:11 monomer-auvidea-1 nvargus-daemon[5276]: SCF: Error Timeout:  (propagating from src/common/Utils.cpp, function workerThread(), line 116)
Sep 16 12:04:11 monomer-auvidea-1 nvargus-daemon[5276]: SCF: Error Timeout: Worker thread IspHw frameComplete failed (in src/common/Utils.cpp, function workerThread(), line 133)
Sep 16 12:04:11 monomer-auvidea-1 nvargus-daemon[5276]: Error: waitCsiFrameStart timeout guid 0
```

Other than this the gstreamer appears to run normally, except no images being displayed on the screen.

I am using the [Auvidea firmware](https://auvidea.eu/firmware/), so suspicious this is conflicting with something in the driver patch. 

Has anyone been able to successfully use the IMX477 with the Auvidea carrier board?

rrcarlosrodriguez, 2020-09-17:
Hi @mattcarp12, 

Using our patches with the Auvidea board is not straightforward, you would need changes on the DTB side in terms of I2C, GPIOs and CSI connections.

If you are interested on hiring RidgeRun to help you adding this support please contact us support@ridgerun.com. You can read more about our services here:
https://developer.ridgerun.com/wiki/index.php?title=RidgeRun_Professional_Services

closed 2020-09-17 by rrcarlosrodriguez

witszym, 2020-11-04:
@mattcarp12 HI, I have the same problem with Auvidea carrier board. Did you resolve this issue ?