RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #22: Can't manage to get it working with v4l2

mdegans, 2020-12-15:
this looks good

```
$ v4l2-ctl --all -d /dev/video0
Driver Info (not using libv4l2):
        Driver name   : tegra-video
        Card type     : vi-output, imx477 7-001a
        Bus info      : platform:54080000.vi:0
        Driver version: 4.9.140
        Capabilities  : 0x84200001
                Video Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 0: no power)
Format Video Capture:
        Width/Height      : 1920/1080
        Pixel Format      : 'RG10'
        Field             : None
        Bytes per Line    : 3840
        Size Image        : 4147200
        Colorspace        : sRGB
        Transfer Function : Default (maps to sRGB)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Full Range)
        Flags             : 

Camera Controls

                     group_hold 0x009a2003 (bool)   : default=0 value=0 flags=execute-on-write
                    sensor_mode 0x009a2008 (int64)  : min=0 max=0 step=0 default=0 value=1 flags=slider
                           gain 0x009a2009 (int64)  : min=0 max=0 step=0 default=0 value=356 flags=slider
                       exposure 0x009a200a (int64)  : min=0 max=0 step=0 default=0 value=33333 flags=slider
                     frame_rate 0x009a200b (int64)  : min=0 max=0 step=0 default=0 value=30000001 flags=slider
                    bypass_mode 0x009a2064 (intmenu): min=0 max=1 default=0 value=1
                override_enable 0x009a2065 (intmenu): min=0 max=1 default=0 value=1
                   height_align 0x009a2066 (int)    : min=1 max=16 step=1 default=1 value=1
                     size_align 0x009a2067 (intmenu): min=0 max=2 default=0 value=0
               write_isp_format 0x009a2068 (bool)   : default=0 value=0
       sensor_signal_properties 0x009a2069 (u32)    : min=0 max=0 step=0 default=0 flags=read-only, has-payload
        sensor_image_properties 0x009a206a (u32)    : min=0 max=0 step=0 default=0 flags=read-only, has-payload
      sensor_control_properties 0x009a206b (u32)    : min=0 max=0 step=0 default=0 flags=read-only, has-payload
              sensor_dv_timings 0x009a206c (u32)    : min=0 max=0 step=0 default=0 flags=read-only, has-payload
               low_latency_mode 0x009a206d (bool)   : default=0 value=0
               preferred_stride 0x009a206e (int)    : min=0 max=65535 step=1 default=0 value=0
                   sensor_modes 0x009a2082 (int)    : min=0 max=30 step=1 default=30 value=2 flags=read-only
```

but when i test it:
```
$ v4l2-ctl -d /dev/video0 --set-fmt-video=width=4032,height=3040,pixelformat=RG10 --set-ctrl bypass_mode=0 --stream-mmap --stream-count=10
```

I get this in dmesg:
```
[ 7882.147255] misc tegra_camera_ctrl: tegra_camera_update_isobw: Warning, Requested ISO BW 1640625 has been capped to VI's max BW 1500000
[ 7882.266796] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 0
[ 7882.300163] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 1
[ 7882.333655] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 2
[ 7882.367046] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 3
[ 7882.400474] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 4
[ 7882.433899] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 5
[ 7882.467586] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 6
[ 7882.500760] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 7
[ 7882.534503] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 8
[ 7882.567789] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 9
[ 7882.602220] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 10
[ 7882.634806] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 11
[ 7882.668284] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 12
[ 7882.701768] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 13
[ 7882.735065] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 14
[ 7882.768576] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 15
[ 7882.802267] vi 54080000.vi: tegra_channel_error_status:error 20022 frame 16
...
```

I found a [forum post](https://forums.developer.nvidia.com/t/why-use-v4l2-ctl-command-get-raw-data-is-alway-zero-at-jetson-tx1-r28-2/60345/4?u=mdegans) referencing this issue pointing to the driver. Any help would be greatly appreciated. I am working on a camera app for a client and have been using v4l2 as opposed to argus. Switching to argus would be a lot of work with our current codebase.

dicksondickson, 2021-01-03:
I am too having this exact same issue.

rrcarlosrodriguez, 2021-01-19:
Hi @dicksondickson and @mdegans,

I just confirmed this issue is no longer present in the updated patches for Jetpack 4.4.1. I ran the exact same pipeline and it worked for me. Please give it a try and let me know how it goes.

mdegans, 2021-01-19:
Thanks, @dicksondickson

I will apply the new patch and let you know.

dicksondickson, 2021-01-20:
Thanks @rrcarlosrodriguez @mdegans .
Does the latest debian packages work with the 2gb Nano?

rrcarlosrodriguez, 2021-01-20:
@dicksondickson I haven't tried to be honest, we don't have that one available. Please test and let me know if it works

dicksondickson, 2021-01-20:
@rrcarlosrodriguez 
I tried installing from the deb package onto the Nano 2gb and it doesn't boot up after install. 
The dtb file is different.

mdegans, 2021-02-08:
@dicksondickson
Got a chance to test and I am still getting dmesg result with the same command with the [latest revision](https://github.com/RidgeRun/NVIDIA-Jetson-IMX477-RPIV3/commit/5803d0d1ffa5273025b7c33c9046b1aa3127523c) as of writing and JetPack 4.5.

Regarding the 2GB Nano, I have one available so I'll test when I get a chance.

lucomsky, 2021-04-24:
I have the same error on 2GB Nano.

falumamx, 2021-06-05:
I have the same problem with JetPack 4.5.1 and Jetson Nano 2GB  😐

vroma1, 2021-11-18:
I have the same problem with JetPack 4.6.1 and Jetson Nano 2GB

dustinkerstein, 2021-11-18:
Check out this thread - https://forums.developer.nvidia.com/t/nano-dev-kit-b01-raspi-hq-cam-doesnt-work-with-v4l2-ctl-jetpack-4-6/190384/26 

It contains a patch if you're willing to recompile the kernel. The fix should also be in the next L4T release.