RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #23: Blinking yellow video

jbarria, 2021-01-11:
Hello guys,

first of all, thank you for this amazing project that allow us using such a great camera with Jetson Nano.

I'm using a RPi HQ camera with Jetson Nano DK in a custom ROV system using this driver and it has been working great in lab environment, but an issue showed up when I did some field tests. As you can see in the attached videos, the video started to blink yellow during 15 - 40 seconds (I edited the videos to extract the clips without exceding the 10MB limit, but I can give the original videos recorded by the Nano if it's necessary). I think it could be related with the sunlight because in both videos it happened when the sun was in the screen. Anyway it was impossible to reproduce it on a lab environment (I tried illuminating the lens of the camera with the flashlight of my cell phone but it didn't happened again).

As context, the GStreamer pipelines I'm using to record are (as a RidgeRun fan I am using GstD and GstInterpipe also in my project):

`nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1920,height=1080,framerate=60/1 ! nvvidconv ! queue ! interpipesink
`
`interpipesrc ! queue ! nvvidconv ! textoverlay ! nvvidconv ! nvv4l2h264enc ! h264parse ! matroskamux ! filesink
`
I know that maybe it's not enough information to make a full diagnostic of what is happening but at least I want to know if you have some idea of where can I check (or maybe some quick solution to test).

Thank you very much in advance for your help.

https://user-images.githubusercontent.com/7014635/104191292-17bc1c00-53fc-11eb-8ca4-536a7e6607a3.mp4

https://user-images.githubusercontent.com/7014635/104191806-c19ba880-53fc-11eb-901f-aa5590f9be91.mp4

rrcarlosrodriguez, 2021-01-19:
Hi @jbarria , I just answered your issue in devtalk, but let's discuss it over here, even better.

As I said, let's see how it behaves disabling the auto exposure and the auto white balance.

## Test 1: aelock=true

Run the pipeline with `nvarguscamerasrc aelock=true`

## Test 1: awblock=true

Run the pipeline with `nvarguscamerasrc awblock=true`

Let me know the result of both tests and we can continue analyzing what is going on

jbarria, 2021-01-21:
Thank you very much for your response and I'm sorry for the delayed response but I wanted to run the tests first.

# Results:

## awblock=true

The problem is still happening. Not any significant change is observed

## aelock=true

The problem stopped happening, the screen just turns full white until the light dissapears:

https://user-images.githubusercontent.com/7014635/105344148-56529300-5bc1-11eb-9b36-5ab2c4fdbd78.mp4

Anyway, when aelock=true and the light conditions change abruptly, visibility becomes quite bad.

## exposurecompensation

You didn't ask me for this test, but I did it anyway because I assumed the problem could be related to exposure.

The problem stop happening when **exposurecompensation >= 1.4** but the video is very brightful.

# Conclusions

After running the tests it looks like the auto-exposure feature is generating this problem. So I guess the solution is to change the **exposuretimerange** property.. am I right? do you think the same? I will try playing with that property but it's a little hard to make it blindly. Do you have any suggestion to follow up?

rrcarlosrodriguez, 2021-02-04:
Hi @jbarria , we found an issue in our exposure range and the exposure calculation. I'm not sure if your issue might be related to that but I think it's worth to give it a try. Let us know if that helps and fixes your issue

jbarria, 2021-02-08:
Hi @rrcarlosrodriguez , I'm sorry but I didn't understand.. what can I try? I still have not been able to fix the problem

jbarria, 2021-02-17:
Hello @rrcarlosrodriguez , do you have any news regarding the exposure calculation issue? maybe a patch or something that let me confirm that my issue is related?