RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #25: Exposure not changing

ori333, 2021-01-14:
Unable to change exposure time.
clip of gstreamer command:
"nvarguscamerasrc wbmode=0 awblock=true aelock=true sensor-id=%d sensor-mode=%d exposuretimerange=\"%d %d\" gainrange=\"%d %d\" ispdigitalgainrange=\"1 1\" aeantibanding=0 ! "

Note: changing gain works.

ori333, 2021-01-15:
After some testing we realized it was not that we can't change exposure, it's that the exposure ranges displayed are not correct:

From: https://forums.developer.nvidia.com/t/raspberry-pi-hq-camera-in-jetson-nano/142762/76
1.
When using gst-launch-1.0 nvarguscamerasrc … it shows that both sensor modes have an exposure range of [13000,683709000]. However, both of those values output a warning ‘invalid time range’, and the camera defaults to some default exposure time.
In:
`gst-launch-1.0 nvarguscamerasrc exposuretimerange="13000 13000" wbmode=0 sensor_id=0 sensor-mode=1 ispdigitalgainrange="1 1" aelock=true gainrange="4 4" ! nvoverlaysink`
Out:
```
GST_ARGUS: NvArgusCameraSrc: Setting Exposure Time Range : 13000 13000
GST_ARGUS: Invalid Exposure Time Range Input
```

2. The values that don’t produce an error are [34000,358733008]. Suspiciously close to the gst-inspect-1.0 nvarguscamerasrc example.

3. For sensor-mode=1: The exposure values from [34000,33333333] actually change the exposure time. However, the range [33333334,358733008] does not produce a warning message and defaults the exposure time to some arbitrary default value which seems to be relatively dim.

[image]

rrcarlosrodriguez, 2021-02-04:
Hi @ori333 , thanks for your report,  this issue was fixed in our latest support for Jetpack 4.5. Please give it a try to the new ranges definition and the new exposure calculation.

closed 2021-02-04 by rrcarlosrodriguez