RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #27: Custom Gamma Curve?

schoolpost, 2021-01-20:
How can you set a custom gamma curve for the processed image?

I see in the camera_overrides.isp options for:
`
tc.v2.user.presets[0].Gamma.Value=2.4000;
tc.v2.user.presets[0].UserCurve.Enable=FALSE;
tc.v2.user.presets[0].UserCurve.Points={ 0.0000, 0.0000, 0.2500, 0.2500, 0.5000, 0.5000, 0.7500, 0.7500, 1.0000, 1.0000 };
tc.v2.user.presets[0].CurveControl.Enable=TRUE;
tc.v2.user.presets[0].CurveControl.AdjustHighlights=0.7500;
tc.v2.user.presets[0].CurveControl.MidtoneBrightness=0.5000;
tc.v2.user.presets[0].CurveControl.MidtoneContrast=0.5000;
tc.v2.user.presets[0].CurveControl.AdjustShadows=0.2500;
tc.v2.user.presets[0].Brightness.Enable=FALSE;
tc.v2.user.presets[0].Brightness.Value=1.0000;
`

I want to set my own points, how many can I do max? and how to enable it? I already try to do UserCurve.Enable but the image looks the same from gstreamer:

`
gst-launch-1.0 nvarguscamerasrc sensor-id=$SENSOR_ID gainrange="1 1" ispdigitalgainrange="1 1" wbmode="daylight" aelock=true ! "video/x-raw(memory:NVMM),width=4032,height=3040,framerate=$FRAMERATE/1" ! nvvidconv flip-method=2 ! "video/x-raw(memory:NVMM),width=4032,height=3040,framerate=$FRAMERATE/1" ! nvoverlaysink
`

mdegans, 2021-02-05:
@schoolpost 

My understanding is Nvidia's software to generate these profiles is private and they only share it with camera manufacturers and companies like leopard imaging. You could try asking on the Nvidia forum.

mdegans, 2021-03-18:
@schoolpost 

So, coming back to this, [libargus lets you do it](https://docs.nvidia.com/jetson/l4t-multimedia/classArgus_1_1IAutoControlSettings.html). It's not a very nice API, however. You're best bet might be to modify `nvarguscamerasrc` to set and get the tonemap curves rather than try to write something from scratch.

DavidSoto, 2021-04-27:
Your understanding is correct in terms of the ISP calibration tools to create a custom calibration file, you would need to contact leopard imaging for a custom file.

closed 2021-04-27 by DavidSoto