RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #30: problem with jetson_io.py with installed IMX477

EricXh, 2021-02-02:
Hello,

I want to start the jetson_io.py tool, but it is not starting correctly.
The window shortly is visible for a fraction of a second and then it dissapers.
No error message.

I tried the init.py hack and the dtb file is from your hack in the /boot/dtb directory.
Could there be a problem in combination with your hack and the jetson_io tool?

JetPack 4.4.1 on B01 board with HQ camera (R8 removed) with all updates (apt update).

BR
Erich

EricXh, 2021-03-04:
This bug is very bad for us, I did a new setop of the linux on nano, changed with jetson_io the SPI output to on and then installed your hack for the IMX477 HQ camera. After that the SPI is not working anymore - so I cannot activate the SPI out when using the HQ camera.
This is essential for out design, this is not a camera board, this is an embedded board where you need the GPIO/I2C/SPI/PWM and so on also, otherwise I could youse a cheap PC, but I need the peripherals in such a design.
Hope you can comment here after 1 month!

BR
Erich

DavidSoto, 2021-03-04:
Hi @EricXh, sorry to hear about the problems that you are experiencing. Please recall that our driver is a free contribution to the community for projects where the driver (and its current status) is suitable, we can't guarantee that it will work for all the possible use cases. 

Now, I don't understand how your question is related to the camera driver. We only use MIPI to get images and i2c to communicate with the camera. If you have SPI problems likely is not related to the driver. However, if you still consider that it might be driver related and you are interested on hiring the RidgeRun engineering team to work in your project, feel free to [contact-us](https://www.ridgerun.com/contact), our support team will provide you pricing information and possible start dates for your project.

closed 2021-03-04 by DavidSoto

EricXh, 2021-03-05:
Ok,
I make a fresh installation on the nano.
I can start jetson_io to set SPI out.
Everything works on SPI.
I install your HQ driver with debian mode/style.
After reboot I can see images from camera, when I start jetson_io it is not working anymore and SPI out ist also not working.
So I think its related to your driver, but I see its "freeware" so its: take it or leave.