RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #32: Don't see /dev/video0 anymore in 4.5

mdegans, 2021-02-05:
System image built using JP 4.5 kernel sources in `public_source.tbz2` with [these nano patches](https://github.com/RidgeRun/NVIDIA-Jetson-IMX477-RPIV3/commit/5803d0d1ffa5273025b7c33c9046b1aa3127523c)

Relevant kernel config bits [full config.gz here](https://github.com/RidgeRun/NVIDIA-Jetson-IMX477-RPIV3/files/5934761/config.gz)
:
```
anzu@test45477:~$ cat /proc/config.gz | gunzip | grep IMX
# CONFIG_TOUCHSCREEN_IMX6UL_TSC is not set
CONFIG_VIDEO_IMX185=m
CONFIG_VIDEO_IMX477=y
CONFIG_VIDEO_IMX219=y
CONFIG_VIDEO_IMX268=y
CONFIG_VIDEO_IMX274=m
CONFIG_VIDEO_IMX318=y
CONFIG_VIDEO_IMX390=y
CONFIG_VIDEO_IMX204=y
```

dmesg bits:
```
anzu@test45477:~$ dmesg | grep imx
[    1.314183] imx219 7-0010: tegracam sensor driver:imx219_v2.0.6
[    1.337561] imx219 7-0010: imx219_board_setup: error during i2c read probe (-121)
[    1.337626] imx219 7-0010: board setup failed
[    1.337678] imx219: probe of 7-0010 failed with error -121
[    1.338091] imx219 8-0010: tegracam sensor driver:imx219_v2.0.6
[    1.361444] imx219 8-0010: imx219_board_setup: error during i2c read probe (-121)
[    1.361501] imx219 8-0010: board setup failed
[    1.361548] imx219: probe of 8-0010 failed with error -121
```

device info:
```
$ cat /proc/device-tree/compatible 
nvidia,p3449-0000-b00+p3448-0000-b00nvidia,jetson-nanonvidia,tegra210
```

[zipped dtb here](https://github.com/RidgeRun/NVIDIA-Jetson-IMX477-RPIV3/files/5934779/tegra210-p3448-0000-p3449-0000-b00.zip)

Any ideas? I have not yet decompiled the dtb and taken a look.

Nitepone, 2021-02-06:
The device tree that you supplied doesn't look to have anything that would enable imx219 probe: but the dmesg log you provided indicates that your active tree has them enabled. Additionally, the phandles for plugin-manager that set imager status look to enable the imx477. This leads me to believe that the device tree you supplied might not be the one active on your Nano.

Can you `cat /proc/device-tree/nvidia,dtbbuildtime` on your Nano? This should help to verify that you are running the device tree that you built.

mdegans, 2021-02-08:
@Nitepone 
Thanks for the tip. The device tree was indeed not loaded. I had not added the `FDT` line to `extlinux.conf`.

closed 2021-02-08 by mdegans

Floritium, 2022-02-07:
Hello there @mdegans @Nitepone ,

I'm having a very similar issue, I'm trying to run a object detection net on my jetson nano 4gb. The CSI cams worked before I installed the neural nets from https://github.com/dusty-nv/jetson-utils but now they don't. I sadly have only been working on the Jetson for a pretty short time and I have no idea what a device tree is or how i would build one. The tutorials are kind of hard to understand for me. Plus I'm worried that since I have a few more python scripts on the Jetson that those might not work with a changed device tree. If you could please point me in the right direction that would be awesome! I sadly also don't understand your last response from Feb 8th 21.

My dmesg output is the same as yours with the exception that i have a imx477 connected.

``` cat /proc/device-tree/compatible ``` also shows the same thing as yours.

and ``` cat /proc/config.gz | gunzip | grep IMX ``` shows this:
```
# CONFIG_TOUCHSCREEN_IMX6UL_TSC is not set
CONFIG_VIDEO_IMX185=m
CONFIG_VIDEO_IMX477=y
CONFIG_VIDEO_IMX219=y
CONFIG_VIDEO_IMX268=y
CONFIG_VIDEO_IMX274=m
CONFIG_VIDEO_IMX318=y
CONFIG_VIDEO_IMX390=y
CONFIG_VIDEO_IMX204=y
# CONFIG_SND_SOC_IMX_AUDMUX is not set
```

Also ``` cat /proc/device-tree/nvidia,dtbbuildtime ``` returns the 29th of November last year so i would assume that I unknowingly changed something in the process of installing the neural net and trying to get it to work so that the cams don't work anymore. If you need anymore info just tell me what and how to get it.

Thanks in advance