RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #33: Any Change we could get a prebuilt driver release for Jetpack 4.5.1

Andrewiski, 2021-04-10:
Any chanse you would be willing to add pre built deb packages for Jetpack 4.5.1?

https://drive.google.com/drive/folders/1bud4kq9eV6eerejPCrelL1upY5VNsRjf?usp=sharing

KamelAbdelouahab, 2021-04-27:
Bump ! 
Is the JP4.5 Debian package compatible with JP4.5.1 ?
Would the kernel rebuild operation detailed [here](https://developer.ridgerun.com/wiki/index.php?title=Raspberry_Pi_HQ_camera_IMX477_Linux_driver_for_Jetson#Download_the_JetPack_sources) work with  JP4.5.1 ? Thanks !

lucomsky, 2022-01-14:
Let's change issue name to «... 4.5.1 - 4.6»!
@rrcarlosrodriguez @dchvs Can you please store the Debian packages' build scripts in this repository?

lucomsky, 2022-01-14:
> Bump ! Is the JP4.5 Debian package compatible with JP4.5.1 ? Would the kernel rebuild operation detailed [here](https://developer.ridgerun.com/wiki/index.php?title=Raspberry_Pi_HQ_camera_IMX477_Linux_driver_for_Jetson#Download_the_JetPack_sources) work with JP4.5.1 ? Thanks !

I have no success with 4.5 on JP4.5.1 (B01):
```luk@linux:~/jetpack-4.5$ sudo apt-get install --reinstall ./nvidia-l4t-kernel_4.9.201-tegra-32.5.0-20210115145440_arm64.deb 
[sudo] password for luk: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'nvidia-l4t-kernel' instead of './nvidia-l4t-kernel_4.9.201-tegra-32.5.0-20210115145440_arm64.deb'
The following packages will be REMOVED:
  nvidia-l4t-kernel-dtbs nvidia-l4t-kernel-headers
The following packages will be DOWNGRADED:
  nvidia-l4t-kernel
0 upgraded, 0 newly installed, 1 downgraded, 2 to remove and 320 not upgraded.
Need to get 0 B/160 MB of archives.
After this operation, 891 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 /home/luk/jetpack-4.5/nvidia-l4t-kernel_4.9.201-tegra-32.5.0-20210115145440_arm64.deb nvidia-l4t-kernel arm64 4.9.201-tegra-32.5.0-20210115145440 [160 MB]
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 177866 files and directories currently installed.)
Removing nvidia-l4t-kernel-dtbs (4.9.201-tegra-32.5.1-20210219084526) ...
Removing nvidia-l4t-kernel-headers (4.9.201-tegra-32.5.1-20210219084526) ...
dpkg: warning: downgrading nvidia-l4t-kernel from 4.9.201-tegra-32.5.1-20210219084526 to 4.9.201-tegra-32.5.0-20210115145440
(Reading database ... 161633 files and directories currently installed.)
Preparing to unpack .../nvidia-l4t-kernel_4.9.201-tegra-32.5.0-20210115145440_arm64.deb ...
Unpacking nvidia-l4t-kernel (4.9.201-tegra-32.5.0-20210115145440) over (4.9.201-tegra-32.5.1-20210219084526) ...
Setting up nvidia-l4t-kernel (4.9.201-tegra-32.5.0-20210115145440) ...
depmod: ERROR: could not open directory /lib/modules/4.9.201-tegra: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
Using the existing boot entry 'primary'
3448-300---1--jetson-nano-devkit-mmcblk0p1
Processing triggers for nvidia-l4t-kernel (4.9.201-tegra-32.5.0-20210115145440) ...

luk@linux:~/jetpack-4.5$ sudo apt-get install --reinstall ./nvidia-l4t-kernel-dtbs_4.9.201-tegra-32.5.0-20210115145440_arm64.deb 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'nvidia-l4t-kernel-dtbs' instead of './nvidia-l4t-kernel-dtbs_4.9.201-tegra-32.5.0-20210115145440_arm64.deb'
The following NEW packages will be installed:
  nvidia-l4t-kernel-dtbs
0 upgraded, 1 newly installed, 0 to remove and 321 not upgraded.
Need to get 0 B/72,1 kB of archives.
After this operation, 2 726 kB of additional disk space will be used.
Get:1 /home/luk/jetpack-4.5/nvidia-l4t-kernel-dtbs_4.9.201-tegra-32.5.0-20210115145440_arm64.deb nvidia-l4t-kernel-dtbs arm64 4.9.201-tegra-32.5.0-20210115145440 [72,1 kB]
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package nvidia-l4t-kernel-dtbs.
(Reading database ... 161633 files and directories currently installed.)
Preparing to unpack .../nvidia-l4t-kernel-dtbs_4.9.201-tegra-32.5.0-20210115145440_arm64.deb ...
Unpacking nvidia-l4t-kernel-dtbs (4.9.201-tegra-32.5.0-20210115145440) ...
Setting up nvidia-l4t-kernel-dtbs (4.9.201-tegra-32.5.0-20210115145440) ...
```

xlla, 2022-07-24:
> I have no success with 4.5 on JP4.5.1 (B01):

same issue here, must I use option B to patch & recompile kernel by myself to use imx477 with jetson nano b01?

it destroy so many work things but left errors here.
```

sudo apt-get install --reinstall ./nvidia-l4t-kernel_4.9.201-tegra-32.5.0-20210115145440_arm64.deb
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'nvidia-l4t-kernel' instead of './nvidia-l4t-kernel_4.9.201-tegra-32.5.0-20210115145440_arm64.deb'
The following packages were automatically installed and are no longer required:
  cython gdal-data gir1.2-gst-plugins-bad-1.0 libarmadillo-dev libarmadillo8 libarpack2 libarpack2-dev libbtf1 libcxsparse3 libdap-dev libdap25
  libdapclient6v5 libdapserver7v5 libepsilon-dev libepsilon1 libflann-dev libflann1.9 libfreexl-dev libfreexl1 libfyba-dev libfyba0 libgdcm2.8
  libgeos-3.6.2 libgeos-c1v5 libgeos-dev libgif-dev libgl2ps-dev libgl2ps1.4 libgraphblas1 libhdf4-0-alt libhdf4-alt-dev libhdf5-mpi-dev
  libhdf5-openmpi-100 libhdf5-openmpi-dev libjs-sphinxdoc libjs-underscore libjson-c-dev libjsoncpp-dev libjsoncpp1 libkml-dev libkmlbase1
  libkmlconvenience1 libkmldom1 libkmlengine1 libkmlregionator1 libkmlxsd1 libldl2 libminizip-dev libminizip1 libnetcdf-c++4
  libnetcdf-cxx-legacy-dev libnetcdf-dev libnetcdf13 libopenni-dev libopenni-sensor-pointclouds0 libopenni0 libopenni2-0 libopenni2-dev
  libpoppler-dev libpoppler-private-dev libpq-dev libpq5 libqhull-dev libqhull-r7 libqhull7 libqt5concurrent5 libqt5designercomponents5
  libqt5opengl5 libqt5quickparticles5 libqt5quicktest5 libqt5quickwidgets5 librhash0 libsocket++1 libsuperlu-dev libsuperlu5 liburiparser-dev
  liburiparser1 libwebp-dev libxerces-c-dev libxerces-c3.2 odbcinst odbcinst1debian2 openni-utils proj-data python-autobahn python-automat
  python-catkin-pkg python-catkin-pkg-modules python-cbor python-click python-colorama python-concurrent.futures python-constantly python-dateutil
  python-hyperlink python-incremental python-lz4 python-mpi4py python-nacl python-openssl python-pam python-pyasn1-modules python-pyparsing
  python-qrcode python-rosdep-modules python-rosdistro python-rosdistro-modules python-rospkg python-rospkg-modules python-service-identity
  python-snappy python-trie python-trollius python-twisted python-twisted-bin python-twisted-core python-txaio python-u-msgpack python-ubjson
  python-wsaccel python-yaml qtbase5-dev-tools ros-melodic-class-loader ros-melodic-cpp-common ros-melodic-ros-environment ros-melodic-rosclean
  ros-melodic-roscpp-serialization ros-melodic-roscpp-traits ros-melodic-rosgraph ros-melodic-roslz4 ros-melodic-rosmaster ros-melodic-rospack
  ros-melodic-rosparam ros-melodic-rostime ros-melodic-xmlrpcpp tcl-dev tcl8.6-dev tk-dev tk8.6-dev unixodbc-dev
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED
  nvidia-l4t-3d-core nvidia-l4t-camera nvidia-l4t-cuda nvidia-l4t-graphics-demos nvidia-l4t-gstreamer nvidia-l4t-jetson-io
  nvidia-l4t-jetson-multimedia-api nvidia-l4t-kernel-dtbs nvidia-l4t-kernel-headers nvidia-l4t-multimedia
The following packages will be DOWNGRADED:
  nvidia-l4t-kernel
0 to upgrade, 0 to newly install, 1 to downgrade, 10 to remove and 398 not to upgrade.
Need to get 0 B/160 MB of archives.
After this operation, 587 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 /home/xlla/nvidia-l4t-kernel_4.9.201-tegra-32.5.0-20210115145440_arm64.deb nvidia-l4t-kernel arm64 4.9.201-tegra-32.5.0-20210115145440 [160 MB]
(Reading database ... 382904 files and directories currently installed.)
Removing nvidia-l4t-graphics-demos (32.5.1-20210519110732) ...
Removing nvidia-l4t-gstreamer (32.5.1-20210519110732) ...
Removing nvidia-l4t-jetson-multimedia-api (32.5.1-20210519110732) ...
Removing nvidia-l4t-camera (32.5.1-20210519110732) ...
Removing nvidia-l4t-multimedia (32.5.1-20210519110732) ...
Removing nvidia-l4t-jetson-io (32.5.1-20210219084526) ...
Removing nvidia-l4t-kernel-dtbs (4.9.201-tegra-32.5.1-20210505093645) ...
Removing nvidia-l4t-kernel-headers (4.9.201-tegra-32.5.1-20210505093645) ...
Removing nvidia-l4t-cuda (32.5.1-20210519110732) ...
Removing nvidia-l4t-3d-core (32.5.1-20210519110732) ...
dpkg: warning: downgrading nvidia-l4t-kernel from 4.9.201-tegra-32.5.1-20210505093645 to 4.9.201-tegra-32.5.0-20210115145440
(Reading database ... 364565 files and directories currently installed.)
Preparing to unpack .../nvidia-l4t-kernel_4.9.201-tegra-32.5.0-20210115145440_arm64.deb ...
Unpacking nvidia-l4t-kernel (4.9.201-tegra-32.5.0-20210115145440) over (4.9.201-tegra-32.5.1-20210505093645) ...
Setting up nvidia-l4t-kernel (4.9.201-tegra-32.5.0-20210115145440) ...
depmod: ERROR: could not open directory /lib/modules/4.9.201-tegra: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
Using the existing boot entry 'primary'
3448-300---1--jetson-nano-devkit-mmcblk0p1
Processing triggers for nvidia-l4t-kernel (4.9.201-tegra-32.5.0-20210115145440) ...
```