RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #36: 4032x3040 @ 60fps ?

Buanderie, 2021-07-23:
Hello,

Thanks for providing this driver for free!
I was wondering if it would be possible to capture 4032x3040 @ 60fps on Jetson Nano.

As far as I understand, CSI-wise it would be possible, by using 4 lanes, but can the sensor actually do it ?
I tried to modify the dts further more, by setting bus-width to 4, and modifying the num_csi_lanes accordingly. I didn't modify the driver (yet).

Trying to capture @30fps with the 4-lane dts setting, all I get is a system freeze (my debug UART is dead btw, so I'm using a ACM gadget on the OTG port, not ideal...).

Is it even possible to achieve such speed ? I ask before investing more time into it.

Thanks in advance.

rrcarlosrodriguez, 2021-07-29:
Hi @Buanderie , it is not possible to use 4 lanes with the Jetson Nano developer kit, because it exposes 2-lanes only in the MIPI CSI-2 camera port. And with 2 lanes only is not possible to reach 60 fps. You would need a custom hardware to expose 4-lanes.

closed 2021-07-29 by rrcarlosrodriguez

Buanderie, 2021-07-30:
Hello,
Thanks for replying. Sorry I forgot to mention that my question was regarding the production module (on a custom carrier board).
Is the VI capable of handling such datarate ?