RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #37: Driver for Jetson TX2 NX

jmadeira-pdm, 2021-08-20:
Hello,

I am currently working on porting this patch for the TX2 NX, but I have a question regarding the files affected by this patch.

Namely I cannot locate the file imx477.c on the TX2 NX kernel files, is this file supplied somewhere else?

dustinkerstein, 2021-09-27:
@jmadeira-pdm, were you able to find these sources and build a working driver for the TX2 NX? Thanks!

Mark-Bezyinski, 2023-08-18:
> @jmadeira-pdm, were you able to find these sources and build a working driver for the TX2 NX? Thanks!

Hello!
Any updates so far?
Could you help me please connect IMX477 to TX2 NX?

Mark

Bra1nsen, 2023-09-08:
Yes we are also interested in using Sony IMX477 Raspberry HQ Cam with TX2 Module. Any updates?