RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #38: Imx477 get blur image

zhenglh33, 2021-09-10:
Hi, we are using imx477 on xaiver NX

When we took video, we found out the image is partially blurred.
[image]

The area covered by strong light is clear(computer on the right lower side and upper part covered by light), while dark area is blurred( left side)

We checked our lens focus is fine, and image quality is better when it’s in normal indoor lighting condition

How can we solve this issue?

Thanks in advance

zhenglh33, 2021-09-10:
[image]
[image]

DavidSoto, 2021-09-14:
Hi, how are you capturing the image? if you are using GStreamer, can you share the pipeline? Things that come to my mind:

a) Is the camera using the autoexposure? I wonder if the sensor is trying to compensate the low light if it overexposed the sensor. 
b) Have you tried setting different exposure values through the V4L2 controls?

We haven't tested the camera in low light conditions but the keyboard and screen of the laptop in the picture seem overexposed.

zhenglh33, 2021-09-14:
Hi,  I used this one
`gst-launch-1.0 \
nvarguscamerasrc sensor-id=0 bufapi-version=FALSE ! tee name=c0 \
c0. ! 'video/x-raw(memory:NVMM),width=1920,height=1080,framerate=60/1' ! nvvidconv flip-method=3 ! queue ! tee name=deep \
deep. ! nvv4l2h264enc bitrate=60000000 ! h264parse ! matroskamux ! filesink location=record_90_flip.mkv \
deep. ! nvegltransform ! nveglglessink sync='FALSE' -e`

(a)The camera sets as autoexposure, (b) I checked the exposure and gain, actually the exposure value did not change, the ae function changed gain value