RidgeRun/NVIDIA-Jetson-IMX477-RPIV3 #45: Jetson Orin Nano Support

vminin2000, 2023-06-30:
I have recently procured a Jetson Orin Nano and I was wondering if you wonderful people would come out with a patch for Orin Nano, or if you have any other advice as to how to fix my kernel/ driver situation so I can use the IMX477 HQ RPI camera. Thanks!

jcaballeros, 2023-06-30:
Have you tried enabling the IMX477 support using jetson-io.py? I believe that it should be supported. You can check how to run jetson-io.py in the following wiki, it was written for the IMX219 but the procedure for IMX477 should be similar:

https://developer.ridgerun.com/wiki/index.php/NVIDIA_Jetson_Orin_Nano/Camera_Sensors_Support/IMX219

lleon95, 2023-10-06:
Hi @bjajoh 

Are you using a fresh install of jetpack on your Jetson Orin Nano using the SDK Manager?

The imx477 driver is available in Jetpack from 5.x onwards. If the kernel has been modified in any way, the configuration script may fail.

BR,
Leon

vminin2000, 2023-10-06:
I only have the option to use IMX219, what code should I use to test the
camera?

lleon95, 2023-10-07:
> I only have the option to use IMX219, what code should I use to test the camera?
> […](#)

Hi @vminin2000 

Are you using a developer kit for the Orin Nano? Additionally, are you installing the OS by using the NVIDIA SDK Manager? NVIDIA has integrated the imx477 driver into the kernel in the newer versions of the Jetpack (a.k.a the base OS of the Jetson).

BR,
Leon

vminin2000, 2023-10-07:
I am using the developer kit and I also use the sdk to flash it. Is there
an extra add on package that needs to be selected when flashing it? When I
connect the camera and cd into the dev/video* folder no camera is selected.
I have seen conflicting code samples to use the camera what do you
recommend? I can reflash a spare Orin nano I have and see if they have more
options for cameras. Thanks for your help!!

alfredpchacon, 2023-10-10:
Hi @bjajoh and @vminin2000 

I hope you're doing very well! Could you bring us a little bit more information about the systems you are using? I will give you a list of the commands you must run to get the information requested, in order to follow up on the issue.

Requested information:
- NVIDIA Orin Nano board model: `cat /sys/firmware/devicetree/base/model `
- Kernel version: ` uname -a`
- Available device trees: `ls /boot && ls /boot/dtb`
- jetson-io.py file directory

vminin2000, 2023-10-13:
***@***.***:~$ cat /sys/firmware/devicetree/base/model
NVIDIA Orin Nano Developer ***@***.***:~$  uname -a
Linux ubuntu 5.10.120-tegra #1 SMP PREEMPT Tue Aug 1 12:32:50 PDT 2023
aarch64 aarch64 aarch64 GNU/Linux
***@***.***:~$ ls /boot && ls /boot/dtb
dtb
extlinux
Image
Image.t19x
Image.t19x.sig
initrd
initrd.img
initrd.img-5.10.120-tegra
initrd.t19x
initrd.t19x.sig
kernel_tegra234-p3767-0003-p3768-0000-a0.dtb
kernel_tegra234-p3767-0003-p3768-0000-a0-user-custom.dtb
tegra194-p2822-camera-dual-imx274-overlay.dtbo
tegra194-p2822-camera-e2832-overlay.dtbo
tegra194-p2822-camera-e3326-overlay.dtbo
tegra194-p2822-camera-e3331-overlay.dtbo
tegra194-p2822-camera-e3333-overlay.dtbo
tegra194-p2822-camera-eCAM130A_CUXVR-overlay.dtbo
tegra194-p2822-camera-imx185-overlay.dtbo
tegra194-p2822-camera-imx390-overlay.dtbo
tegra194-p2888-0001-p2822-0000-adafruit-sph0645lm4h.dtbo
tegra194-p2888-0001-p2822-0000-adafruit-uda1334a.dtbo
tegra194-p2888-0001-p2822-0000-csi.dtbo
tegra194-p2888-0001-p2822-0000.dtb
tegra194-p2888-0001-p2822-0000.dtb.sig
tegra194-p2888-0001-p2822-0000-fe-pi-audio.dtbo
tegra194-p2888-0001-p2822-0000-hdr40.dtbo
tegra194-p2888-0001-p2822-0000-m2ke.dtbo
tegra194-p2888-0001-p2822-0000-maxn.dtb
tegra194-p2888-0001-p2822-0000-maxn.dtb.sig
tegra194-p2888-0001-p2822-0000-overlay.dtbo
tegra194-p2888-0001-p2822-0000-respeaker-4-mic-array.dtbo
tegra194-p2888-0001-p2822-0000-respeaker-4-mic-lin-array.dtbo
tegra194-p2888-0004-e3900-0000-adafruit-sph0645lm4h.dtbo
tegra194-p2888-0004-e3900-0000-adafruit-uda1334a.dtbo
tegra194-p2888-0004-e3900-0000-csi.dtbo
tegra194-p2888-0004-e3900-0000.dtb
tegra194-p2888-0004-e3900-0000.dtb.sig
tegra194-p2888-0004-e3900-0000-dual-imx274.dtbo
tegra194-p2888-0004-e3900-0000-fe-pi-audio.dtbo
tegra194-p2888-0004-e3900-0000-hdr40.dtbo
tegra194-p2888-0004-e3900-0000-imx274.dtbo
tegra194-p2888-0004-e3900-0000-respeaker-4-mic-array.dtbo
tegra194-p2888-0004-e3900-0000-respeaker-4-mic-lin-array.dtbo
tegra194-p2888-0008-p2822-0000.dtb
tegra194-p2888-0008-p2822-0000.dtb.sig
tegra194-p2888-0008-p2822-0000-maxn.dtb
tegra194-p2888-0008-p2822-0000-overlay.dtbo
tegra194-p3668-0000-p3509-0000.dtb
tegra194-p3668-0000-p3509-0000.dtb.sig
tegra194-p3668-0001-p3509-0000.dtb
tegra194-p3668-0001-p3509-0000.dtb.sig
tegra194-p3668-all-p3509-0000-adafruit-sph0645lm4h.dtbo
tegra194-p3668-all-p3509-0000-adafruit-uda1334a.dtbo
tegra194-p3668-all-p3509-0000-camera-imx219-dual.dtbo
tegra194-p3668-all-p3509-0000-camera-imx477-dual.dtbo
tegra194-p3668-all-p3509-0000-camera-imx477-imx219.dtbo
tegra194-p3668-all-p3509-0000.dtb
tegra194-p3668-all-p3509-0000.dtb.sig
tegra194-p3668-all-p3509-0000-fe-pi-audio.dtbo
tegra194-p3668-all-p3509-0000-hdr40.dtbo
tegra194-p3668-all-p3509-0000-m2ke.dtbo
tegra194-p3668-all-p3509-0000-respeaker-4-mic-array.dtbo
tegra194-p3668-all-p3509-0000-respeaker-4-mic-lin-array.dtbo
tegra194-p3668-p3509-overlay.dtbo
tegra234-p3701-0000-as-p3767-0000-p3737-0000.dtb
tegra234-p3701-0000-as-p3767-0001-p3737-0000.dtb
tegra234-p3701-0000-as-p3767-0003-p3737-0000.dtb
tegra234-p3701-0000-as-p3767-0004-p3737-0000.dtb
tegra234-p3701-0000-p3737-0000.dtb
tegra234-p3701-0000-p3737-0000-kexec.dtb
tegra234-p3701-0002-p3711-0000.dtb
tegra234-p3701-0002-p3740-0002-b01.dtb
tegra234-p3701-0002-p3740-0002.dtb
tegra234-p3701-0004-p3737-0000.dtb
tegra234-p3701-0005-p3737-0000.dtb
tegra234-p3701-0008-p3737-0000.dtb
tegra234-p3701-0008-p3740-0002-c01.dtb
tegra234-p3701-overlay.dtbo
tegra234-p3737-0000-p3701-0000-adafruit-sph0645lm4h.dtbo
tegra234-p3737-0000-p3701-0000-adafruit-uda1334a.dtbo
tegra234-p3737-0000-p3701-0000-csi.dtbo
tegra234-p3737-0000-p3701-0000-fe-pi-audio.dtbo
tegra234-p3737-0000-p3701-0000-hdr40.dtbo
tegra234-p3737-0000-p3701-0000-m2ke.dtbo
tegra234-p3737-0000-p3701-0000-respeaker-4-mic-array.dtbo
tegra234-p3737-0000-p3701-0000-respeaker-4-mic-lin-array.dtbo
tegra234-p3737-audio-codec-rt5658-40pin.dtbo
tegra234-p3737-camera-dual-hawk-ar0234-e3653-overlay.dtbo
tegra234-p3737-camera-dual-imx274-overlay.dtbo
tegra234-p3737-camera-e3331-overlay.dtbo
tegra234-p3737-camera-e3333-overlay.dtbo
tegra234-p3737-camera-hawk-owl-overlay.dtbo
tegra234-p3737-camera-imx185-overlay.dtbo
tegra234-p3737-camera-imx390-overlay.dtbo
tegra234-p3737-overlay.dtbo
tegra234-p3740-camera-hawk-owl-overlay.dtbo
tegra234-p3740-camera-p3785-overlay.dtbo
tegra234-p3767-0000-p3509-a02-adafruit-sph0645lm4h.dtbo
tegra234-p3767-0000-p3509-a02-adafruit-uda1334a.dtbo
tegra234-p3767-0000-p3509-a02-csi.dtbo
tegra234-p3767-0000-p3509-a02-fe-pi-audio.dtbo
tegra234-p3767-0000-p3509-a02-hdr40.dtbo
tegra234-p3767-0000-p3509-a02-m2ke.dtbo
tegra234-p3767-0000-p3509-a02-respeaker-4-mic-array.dtbo
tegra234-p3767-0000-p3509-a02-respeaker-4-mic-lin-array.dtbo
tegra234-p3767-0000-p3768-0000-csi.dtbo
tegra234-p3767-camera-imx219-dual.dtbo
tegra234-p3767-camera-imx477-dual.dtbo
tegra234-p3767-camera-imx477-imx219.dtbo
tegra234-p3767-camera-p3768-imx219-dual.dtbo
tegra234-p3767-camera-p3768-imx477-dual-4lane.dtbo
tegra234-p3767-camera-p3768-imx477-dual.dtbo
tegra234-p3767-camera-p3768-imx477-imx219.dtbo
tegra234-p3767-overlay.dtbo
tegra234-p3768-overlay.dtbo
kernel_tegra234-p3767-0003-p3768-0000-a0.dtb
sudo /opt/nvidia/jetson-io/jetson-io.py # THIS IS THE jetson-io.py directory
It appears as if I do have the IMX 477 camera kernel. How do I test it out
while the camera is connected.Thank you for your help!

ImTheMan153, 2024-03-20:
> […](#)
> -- -- Victor Minin

Hello,
Did you find a solution to use CSI camera with jetson orin nano on jetpack 6? I have an issue which reports:
```

[video_source-1] nvbuf_utils: dmabuf_fd -1 mapped entry NOT found
[video_source-1] Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadExecute:694 NvBufSurfaceFromFd Failed.
[video_source-1] Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, threadFunction:247 (propagating)
```